espressif/idf-component-manager #86: why idf.py build connects to components-file.espressif.com? (IDFGH-14829)

esp-dev, 2025-03-12:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

No any IP requests allows to track users.

### What is the actual behavior?

why idf.py build connects to components-file.espressif.com it doesn't seem safe especially since you can track the calls to this command and what IP address they come from.

### Steps to reproduce.

just call idf.py build and use wireshark

### Build or installation Logs.

```plain

```

esp-dev, 2025-03-14:
Hi @hfudev,

Yes, I use esp-mdns, this is reason. Yesterday or the day before yesterday he did not have access to the Internet and reconstruction was not available.

hfudev, 2025-03-17:
@esp-dev Looks like the issue is resolved, so I'll go ahead and close it. Feel free to report any issues anytime!

closed 2025-03-17 by hfudev