espressif/idf-component-manager #87: Unable to run `. ./export.sh` when installing esp-idf due to error, preventing builds (IDFGH-15146)

abirw, 2025-04-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.5-dev-3190-gabe7680142

### Operating System used.

macOS

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

following this step of the getting started prerequisites: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html#step-4-set-up-the-environment-variables

Have followed all of the MacOS steps up until this, using the default tools install path. Would expect no error in the output and able to build a project afterwards, I get this same error if i try to build as the output suggests. I think the error is coming from dependency resolution issues.

### What is the actual behavior?

produces this error output:
```
aw@awmbp esp-idf % . ./export.sh
Checking "python3" ...
Python 3.9.0
"python3" has been detected
Activating ESP-IDF 5.5
Setting IDF_PATH to '/Users/aw/esp/esp-idf'.
* Checking python version ... 3.9.0
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... zsh
* Detecting outdated tools in system ... OK - no outdated tools found
* Shell completion ... Autocompletion code generated
Traceback (most recent call last):
  File "/Users/aw/esp/esp-idf/tools/idf.py", line 46, in <module>
    from idf_component_manager import idf_extensions
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_manager/idf_extensions.py", line 6, in <module>
    from idf_component_manager.cli.validations import (
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/__init__.py", line 3, in <module>
    from .core import safe_cli
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/core.py", line 17, in <module>
    from .component import init_component
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/component.py", line 6, in <module>
    from idf_component_manager.cli.validations import (
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/validations.py", line 9, in <module>
    from idf_component_manager.core_utils import COMPONENT_FULL_NAME_WITH_SPEC_REGEX
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_manager/core_utils.py", line 21, in <module>
    from idf_component_tools.manifest import Manifest
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/__init__.py", line 10, in <module>
    from .metadata import Metadata
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/metadata.py", line 7, in <module>
    from .schemas import flatten_manifest_file_keys, serialize_list_of_list_of_strings
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/schemas.py", line 14, in <module>
    MANIFEST_JSON_SCHEMA = Manifest.model_json_schema(by_alias=True)
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/pydantic/main.py", line 555, in model_json_schema
    return model_json_schema(
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 2404, in model_json_schema
    schema_generator_instance = schema_generator(by_alias=by_alias, ref_template=ref_template)
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 278, in __init__
    self._schema_type_to_method = self.build_schema_type_to_method()
  File "/Users/aw/.espressif/python_env/idf5.5_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 317, in build_schema_type_to_method
    method_name = f'{key.replace("-", "_")}_schema'
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/typing.py", line 647, in __getattr__
    return getattr(self.__origin__, attr)
AttributeError: '_SpecialForm' object has no attribute 'replace'

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
```

I cannot compile ESP-IDF projects like it suggests

### Steps to reproduce.

1. verify that python3, cmake, ninja and dfu-util are installed using brew
2. `mkdir -p ~/esp`
3. `cd ~/esp`
4. `git clone --recursive git@github.com:espressif/esp-idf.git`
5. `cd ~/esp/esp-id`
6. `./install.sh esp32`
7. `. ./export.sh`

### Build or installation Logs.

```plain

```

abirw, 2025-04-30:
Hey, sorry for the late reply, here you go!

```
Package                   Version
------------------------- ---------
annotated-types           0.7.0
argcomplete               3.6.2
bitarray                  3.3.1
bitstring                 4.3.1
certifi                   2025.1.31
cffi                      1.17.1
charset-normalizer        3.4.1
click                     8.1.8
colorama                  0.4.6
construct                 2.10.70
cryptography              43.0.3
ecdsa                     0.19.1
esp-coredump              1.13.1
esp-idf-diag              0.1.1
esp-idf-kconfig           2.5.0
esp-idf-monitor           1.6.2
esp_idf_nvs_partition_gen 0.1.6
esp_idf_panic_decoder     1.3.0
esp-idf-size              1.6.1
esptool                   4.9.dev6
freertos-gdb              1.0.4
idf-component-manager     2.1.2
idna                      3.10
intelhex                  2.3.0
jsonref                   1.1.0
markdown-it-py            3.0.0
mdurl                     0.1.2
packaging                 25.0
pip                       25.0.1
psutil                    7.0.0
pyclang                   0.6.0
pycparser                 2.22
pydantic                  2.11.3
pydantic_core             2.33.1
pydantic-settings         2.8.1
pyelftools                0.32
pygdbmi                   0.11.0.0
Pygments                  2.19.1
pyparsing                 3.2.3
pyserial                  3.5
python-dotenv             1.1.0
PyYAML                    6.0.2
reedsolo                  1.7.0
requests                  2.32.3
requests-file             2.1.0
requests-toolbelt         1.0.0
rich                      14.0.0
ruamel.yaml               0.18.10
ruamel.yaml.clib          0.2.12
setuptools                71.0.0
six                       1.17.0
tqdm                      4.67.1
typing_extensions         4.13.2
typing-inspection         0.4.0
urllib3                   1.26.20
```

XDanielPaul, 2025-05-05:
MacOS Sequoia 15.4.1
Python 3.9.21

I was unable to reproduce the issue with Python 3.9.21. It could potentially be related to the version of Python being used, and it might have been resolved in later versions. Would you mind trying a newer 3.9.x version to see if the issue persists?

Indastri, 2025-06-02:
Since there was no response from the user - closing. If you will encounter the issue again or any other issue. Feel free to reopen this or create a new issue.

Cheers!

closed 2025-06-02 by Indastri