espressif/idf-component-manager #89: cannot get past '. $HOME/Src/esp/esp-idf/export.sh' without errors (IDFGH-15510)

mlshort, 2025-06-30:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v6.0-dev-527-gadcbdd7da4

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Complete without errors.

### What is the actual behavior?

* Shell completion ... Autocompletion code generated
Traceback (most recent call last):
  File "/home/blackboard/Src/esp/esp-idf/tools/idf.py", line 46, in <module>
    from idf_component_manager import idf_extensions
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/idf_extensions.py", line 6, in <module>
    from idf_component_manager.cli.validations import (
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/__init__.py", line 3, in <module>
    from .core import safe_cli
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/core.py", line 17, in <module>
    from .component import init_component
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/component.py", line 6, in <module>
    from idf_component_manager.cli.validations import (
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/validations.py", line 9, in <module>
    from idf_component_manager.core_utils import COMPONENT_FULL_NAME_WITH_SPEC_REGEX
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/core_utils.py", line 17, in <module>
    from idf_component_tools.manifest import Manifest
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/__init__.py", line 10, in <module>
    from .metadata import Metadata
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/metadata.py", line 7, in <module>
    from .schemas import flatten_manifest_file_keys, serialize_list_of_list_of_strings
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/schemas.py", line 14, in <module>
    MANIFEST_JSON_SCHEMA = Manifest.model_json_schema(by_alias=True)
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/pydantic/main.py", line 555, in model_json_schema
    return model_json_schema(
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 2404, in model_json_schema
    schema_generator_instance = schema_generator(by_alias=by_alias, ref_template=ref_template)
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 278, in __init__
    self._schema_type_to_method = self.build_schema_type_to_method()
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 317, in build_schema_type_to_method
    method_name = f'{key.replace("-", "_")}_schema'
  File "/usr/local/lib/python3.9/typing.py", line 647, in __getattr__
    return getattr(self.__origin__, attr)
AttributeError: '_SpecialForm' object has no attribute 'replace'

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
blackboard@ubuntu:~$

### Steps to reproduce.

mkdir -p ~/Src/esp
cd ~/Src/esp
git clone --recursive https://github.com/espressif/esp-idf.git
cd ~/Scr/esp/esp-idf
./install.sh esp32
. ./export.sh

### Build or installation Logs.

```plain

```

### Diagnostic report archive.

blackboard@ubuntu:~/Src/esp/hello_world$ idf.py diag
Traceback (most recent call last):
  File "/home/blackboard/Src/esp/esp-idf/tools/idf.py", line 46, in <module>
    from idf_component_manager import idf_extensions
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/idf_extensions.py", line 6, in <module>
    from idf_component_manager.cli.validations import (
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/__init__.py", line 3, in <module>
    from .core import safe_cli
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/core.py", line 17, in <module>
    from .component import init_component
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/component.py", line 6, in <module>
    from idf_component_manager.cli.validations import (
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/cli/validations.py", line 9, in <module>
    from idf_component_manager.core_utils import COMPONENT_FULL_NAME_WITH_SPEC_REGEX
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_manager/core_utils.py", line 17, in <module>
    from idf_component_tools.manifest import Manifest
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/__init__.py", line 10, in <module>
    from .metadata import Metadata
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/metadata.py", line 7, in <module>
    from .schemas import flatten_manifest_file_keys, serialize_list_of_list_of_strings
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/idf_component_tools/manifest/schemas.py", line 14, in <module>
    MANIFEST_JSON_SCHEMA = Manifest.model_json_schema(by_alias=True)
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/pydantic/main.py", line 555, in model_json_schema
    return model_json_schema(
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 2404, in model_json_schema
    schema_generator_instance = schema_generator(by_alias=by_alias, ref_template=ref_template)
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 278, in __init__
    self._schema_type_to_method = self.build_schema_type_to_method()
  File "/home/blackboard/.espressif/python_env/idf6.0_py3.9_env/lib/python3.9/site-packages/pydantic/json_schema.py", line 317, in build_schema_type_to_method
    method_name = f'{key.replace("-", "_")}_schema'
  File "/usr/local/lib/python3.9/typing.py", line 647, in __getattr__
    return getattr(self.__origin__, attr)
AttributeError: '_SpecialForm' object has no attribute 'replace'
blackboard@ubuntu:~/Src/esp/hello_world$ ls

### More Information.

Python 3.9.0

Linux ubuntu 4.9.0-xilinx-v2017.1+ espressif/esp-idf#1 SMP PREEMPT Sun Feb 17 18:07:40 MST 2019 armv7l armv7l armv7l GNU/Linux

Distributor ID: Ubuntu
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic

nopnop2002, 2025-06-30:
>But for this platform it requires me to download and build from the source as there is no pre-built python3.9.x distribution for ubuntu-bionic-beaver (arm).

What platform do you use?

mlshort, 2025-08-14:
Kind of you for the update.  I have nothing to substantial a different conclusion and am open to believe it is an issue with 3.9.0.

Unfortunately I have had to move on from this project and was not able to utilize IDF.  :(