espressif/idf-component-manager #91: Ability to self-host a local mirror

ftab, 2025-07-08:
### Related area

Registry

### Is your feature request related to a problem?

I'm currently experiencing an internet outage that has brought our ability to build ESP projects to a halt.

### Describe the solution you'd like

I'd like software and instructions on self-hosting the registry, and corresponding configuration in the idf component to declare a backup registry URL in case the main registry is unavailable (such as during an internet connection outage).

### Describe alternatives you've considered

Git submodules are able to be pushed to a local git repository and cloned locally. I was considering moving all dependencies to submodules to make it possible to build entirely locally, but the dependency tree grows many branches and makes this less practical to move to submodules

### I have checked existing list of Feature requests and I have provided enough information.

- [x] I confirm I have checked existing list of Feature requests and I have provided enough information.

ftab, 2025-07-08:
Ideal use case, an automatically synchronizable mirror so that I can easily update my copy when the internet is working.

ftab, 2025-07-09:
My ultimate goal is a local network mirror of the entire component registry, automatically synchronizing all components periodically, that my GitLab CI runner can access. Is that feasible with this? Maybe if there's an easily processable list of components so I can do a script to sync all.

In trying to get it to work on my local machine, I've tried the following:

For each component: `compote registry sync --component component_name ~/esp-component-mirror`

Note: The `--project-dir` option does not seem to work, it immediately returns having done nothing.

```
$ cd ~/path-to-my-project
$ compote registry sync --project-dir . ~/esp-component-mirror/
NOTICE: Collecting local storage from folder "/home/build/esp-component-mirror"
NOTICE: 1 components loaded from "/home/build/esp-component-mirror" folder
Collecting required components: 0
ERROR: No component need to be downloaded with the specified requirements
```

Uploading the contents of ~/esp-component-mirror to a file server on my local network (`*.apps.mydomain.com` resolves to our in-house Kubernetes cluster and I've set up a deployment on it to serve these files)

Setting registry_url in main/idf_component.yaml to the URL of this file server:
```yaml
dependencies:
  registry_url: https://esp-component-mirror.apps.mydomain.com
  espressif/esp-dsp:
    version: "*"
...
```
or
```yaml
dependencies:
  espressif/esp-dsp:
    registry_url: https://esp-component-mirror.apps.mydomain.com
    version: "*"
...
```

No matter what I try, it doesn't ever hit my server, and dependencies.lock shows registry_url for all of the downloaded components is still components.espressif.com.

I also set ~/.espressif/idf_component_manager.yaml:
```yaml
profiles:
  default:
    local_storage_url:
      - https://esp-component-mirror.apps.mydomain.com
```

`compote version` reports 2.1.2 so it should work according to the documentation.

I've tried `compote cache clear` and `idf.py fullclean` and `rm dependencies.lock` between attempts, to ensure that I am not caching anything and it should behave like my GitLab CI build will.

Getting the component manager globally configured to use it would be the best if I can figure out what I'm doing wrong.

kumekay, 2025-07-10:
> For each component: compote registry sync --component component_name ~/esp-component-mirror

You can supply multiple `--component` arguments to the command, also all the dependencies required for the component will be added to the mirror.

> Note: The --project-dir option does not seem to work, it immediately returns having done nothing.

`--project-dir` option is supposed to point to an existing project, it will check all the components already used by the project (by recursively scanning all idf_component.yml files in the directory) and will create a local mirror for these components. If you specify an empty directory, it will do nothing.

It's possible to make a full mirror of the registry, but it will take a lot of space, because all the versions of all components will be synchronized. Here is a docker compose file with example of self-hosted mirror: https://gist.github.com/kumekay/ca35a4adec84538e8c4929665405efed

Then you should use `local_storage_url` in ~/.espressif/idf_component_manager.yaml is probably the best way to use with local mirror.

kumekay, 2025-07-14:
@ftab Did you have a chance to try mirror options mentioned above?

> compote version reports 2.1.2 so it should work according to the documentation.

I didn't notice it before, but the latest version of the component manager is 2.2.2, and in 2.2 we fixed some issues connected to partial mirrors.

ftab, 2025-07-14:
I got it working. I was saving the config as idf_component_manager.yaml and it was looking for idf_component_manager.yml

Thanks!

closed 2025-07-14 by ftab