espressif/idf-component-manager #92: ESP Component Registry - Partial Service Outage 10.07.2025

kumekay, 2025-07-10:
### The Component Manager version

-

### ESP-IDF Version

-

### python Version

-

### Operating System

-

### Description

# ESP Component Registry - Service Outage

**Status:** 🔴 Partial Outage  
**ETA:** July 11, 2025, 08:00 UTC

## Affected Services
- Component uploads (new versions)
- Search functionality  
- Login of users
- Web ZIP downloads

## Working Services
- Build-time downloads of components ✅
- Pages of individual components and examples ✅
- Existing documentation ✅

### To Reproduce

```cpp
-
```

### Expected behaviour

-

### I have checked existing issues and online Documentation

- [x] I confirm I have checked existing issues and online Documentation.

kumekay, 2025-07-11:
Incident solved. All systems are working as expected.

closed 2025-07-11 by kumekay

Masterpc96, 2025-07-12:
https://components.espressif.com -> gateway timeout

MacWyznawca, 2025-07-12:
Still don't working 

[image]

winston432, 2025-07-12:
still not working

[image]

kumekay, 2025-07-14:
@Masterpc96  @MacWyznawca @winston432 Thank you for comments! On Saturday, we have 2 episodes of outages connected to unusual (and likely malicious) traffic spikes. We are now adding DDoS protection to the web interface and adjusting service autoscaling, to be sure it won't repeat in future.